RoboVerseOrg/RoboVerse #32: inquiry about the mocap teleoperation

Snow-02, 2025-04-07:
Hello,

I’ve read your paper in which you mentioned using a camera as the motion capture input for teleoperation. The concept is fascinating, but I couldn’t find detailed implementation steps in the paper. I’d really appreciate any additional guidance or references you can provide. Below are some specific questions:

1. **Hardware & Software Environment**  
   - Which camera or camera system did you use, and did you require any specific resolution, frame rate, or lens parameters to achieve accurate motion capture?  
   - Regarding the software environment (e.g., operating system, drivers, dependencies), are there particular configurations or setups you recommend?

2. **Motion Capture Workflow**  
   - How did you perform pose estimation or keypoint detection once you had the camera feed? Did you use any existing libraries or algorithms (e.g., OpenPose, MediaPipe, or a custom deep learning model)?  
   - Could you outline the main steps or data flow from capturing frames to extracting pose/position data?

3. **Integration with the Control System**  
   - The paper mentions using camera data as a MoCap input for teleoperation. How does this interface or communicate with your back-end control system or robot (e.g., a manipulator)?  
   - Do you offer an API or any communication protocol that allows integration with existing robot control software?

4. **Code & Examples**  
   - Is there any publicly available example code or documentation that demonstrates how to set up and run this camera-based MoCap system?  
   - If not fully open-source, could you share a brief overview or a diagram of the system’s workflow?

Thank you in advance for any information or pointers you can provide. Your research is very helpful, and I’m eager to learn more about the practical aspects of your approach.

Looking forward to your response!

geng-haoran, 2025-04-09:
For motion capture, we use [Leap Motion](https://www.ultraleap.com/) for teleoperation. Currently, it is only supported in a deprecated version of Metasim, but we plan to integrate the code into the main release version soon!

Fisher-Wang, 2025-04-13:
Please feel free to reopen if you encounter further issues.

closed 2025-04-13 by Fisher-Wang