RoboVerseOrg/RoboVerse #37: Load glb meshes

dementrock, 2025-04-07:
Say I have a glb mesh file. What's the recommended way to load this into roboverse?

More specifically I'm trying to migrate some maniskill environment components, e.g. https://github.com/haosulab/ManiSkill/blob/b1b94d6f341daa8013255a046ccda27719204c8f/mani_skill/utils/scene_builder/kitchen_counter/scene_builder.py#L21

Fisher-Wang, 2025-04-08:
Hi Rocky!

If you are working on IsaacLab, it support [import glb/gltf file](https://isaac-sim.github.io/IsaacLab/main/source/how-to/import_new_asset.html#using-mesh-importer). The recommended workflow for IsaacLab is to first convert your .glb file to USD format and then use the USD file with MetaSim.

Alternatively, you could: 
1. convert the glb file into obj file, and 
2. wrap the obj as urdf (you can refer to [this file](https://github.com/RoboVerseOrg/RoboVerse/blob/6d691fc139bf8097b45d60046b4115fbe7cf7e7a/get_started/example_assets/bbq_sauce/urdf/bbq_sauce.urdf)) or convert it to mjcf (you can use this tool: https://github.com/kevinzakka/obj2mjcf). 

But I'm not sure if step 1 will lose some information.

Please let me know if this solve your question!

dementrock, 2025-04-08:
Thanks @Fisher-Wang ! Is there a solution that works across more simulators, rather than just isaaclab? My hope of using roboverse is to be able to use a faster-launching simulator e.g. mujoco for development and switch to isaaclab for actual large scale training, but right now setting up assets to be cross-simulator seems pretty tedious

Fisher-Wang, 2025-04-08:
I see your point, Rocky! Maybe we can automate the above workflow (as a part of asset conversion) in run time. In that way, users could seamlessly switch between simulators no matter what format of assets is provided.

@charlietcheng, could you share your thoughts on the level of effort required to integrate the asset conversion workflow into runtime?

Fisher-Wang, 2025-04-08:
Related to #45

closed 2025-04-08 by Fisher-Wang