RoboVerseOrg/RoboVerse #38: How to control simulation dt?

dementrock, 2025-04-08:
Didn't find this setting anywhere

charlietcheng, 2025-04-08:
Hi Rocky. Simulation dt is defined as `decimation` inside the task config files. Decimation refers to how many action steps per environment steps in our settings. checkout:

[https://github.com/RoboVerseOrg/RoboVerse/blob/60db826008b7a5db252dfb60c5dd038318390831/metasim/cfg/scenario.py#L62](default scenario config)
[https://github.com/RoboVerseOrg/RoboVerse/blob/60db826008b7a5db252dfb60c5dd038318390831/metasim/cfg/tasks/metaworld/drawer_open_cfg.py#L11](open drawer config)

I hope this helps. Please let me know if you have more questions

dementrock, 2025-04-08:
Thanks @Fisher-Wang ! decimation seems like an integer config and it's not clear how it's mapped to the overall simulation step time in the unit of seconds?

Fisher-Wang, 2025-05-12:
Thanks Rocky! This is resolved now. You can change the dt by setting `scenario.sim_params.dt`

closed 2025-05-12 by Fisher-Wang