RoboVerseOrg/RoboVerse #40: Disabling rendering in HumanoidBench tasks

younggyoseo, 2025-04-08:
Hi all, congrats on great work. I was able to launch some HumanoidBench tasks based on IsaacLab in my machine with Haoran's help. But I noticed that rendering is turned on by default -- is there a way to launch IsaacLab environments without rendering? Would this make experiments more faster if I don't need RGB/Depth observations for my experiments?

younggyoseo, 2025-04-08:
Thank you. I'm finding that fps for HumanoidBench tasks is really low so I guess this should be very important. Looking forward to the update!

Fisher-Wang, 2025-04-08:
Yes It is because we split the state tensors into nested list and dict in `get_states()` method for we want to unify the format. We will try to improve this by
1. using a `query` argument to return only the queried states.
2. using a more efficient data structure (maybe a python class with a whole state tensor and can be lazily queried) instead of nested dict and list
3. profiling to verify the efficiency

EmptyBlueBox, 2025-04-15:
Rendering has been disabled in HumanoidBench, alongside a lot of other efficiency enhancements, leading to a substantial boost in FPS!

https://github.com/RoboVerseOrg/RoboVerse/blob/main/roboverse_learn/humanoidbench_rl/train_sb3.py#L84-L85

closed 2025-04-15 by EmptyBlueBox