RoboVerseOrg/RoboVerse #48: humanoidbench error

WeisonWEileen, 2025-04-08:
running command:
`python roboverse_learn/humanoidbench_rl/train_sb3.py --sim mujoco --num_envs 1 --robot=h1 --task humanoidbench:Stand`
error:
```
WARNING:absl:XML contains a 'NaN'. Please check it carefully.
                    INFO     Reading trajectory using v2 data format                                                                                                                           demo_util.py:31
                    ERROR    No actions found in the trajectory data                                                                                                                        demo_util_v2.py:59
                    ERROR    No states found in the trajectory data                                                                                                                         demo_util_v2.py:71
Traceback (most recent call last):
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/train_sb3.py", line 332, in <module>
    main()
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/train_sb3.py", line 92, in main
    env = Sb3EnvWrapperMujoco(scenario=scenario)
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/sb3_wrapper_mujoco.py", line 40, in __init__
    initial_obs, _ = self.reset()
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/sb3_wrapper_mujoco.py", line 55, in reset
    observation = self.get_humanoid_observation()
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/sb3_wrapper_mujoco.py", line 88, in get_humanoid_observation
    gym_observation = self.env.handler.task.humanoid_obs_flatten_func(envstates)
  File "/home/panwei/RoboVerse/metasim/cfg/tasks/humanoidbench/base_cfg.py", line 72, in humanoid_obs_flatten_func
    env_obs.append(np.concatenate([arr for arr in flattened]))
ValueError: need at least one array to concatenate
```
I checked the `.pkl` dataset file, it was mismatch between in dataset format and code in `metasim/cfg/tasks/humanoidbench/base_cfg.py` when I modified the code as
```
@staticmethod
    def humanoid_obs_flatten_func(envstates: list[EnvState]) -> torch.Tensor:
        """Observation function for humanoid tasks."""
        env_obs = []
        for envstate in envstates:
            flattened = []
            for obj_name, obj_state in envstate.items():
                if obj_name == "h1" or obj_name == "g1" or obj_name == "h1_simple_hand" or obj_name == "h1_hand":
                    for key in ["pos", "rot", "vel", "ang_vel"]:
                        if key in obj_state and obj_state[key] is not None:
                            flattened.append(np.array(obj_state[key]).flatten())
                    if "dof_pos" in obj_state:
                        for key, value in obj_state["dof_pos"].items():
                            if isinstance(value, (np.ndarray, float)):
                                flattened.append(np.array(value).flatten())
                    if "dof_vel" in obj_state:
                        for key, value in obj_state["dof_vel"].items():
                            if isinstance(value, (np.ndarray, float)):
                                flattened.append(np.array(value).flatten())
                elif not obj_name.startswith("metasim"):
                    # flatten pos, rot, vel, ang_vel for other objects. (state-based observation)
                    for key in ["pos", "rot", "vel", "ang_vel"]:
                        if key in obj_state and obj_state[key] is not None:
                            flattened.append(obj_state[key].numpy())
            env_obs.append(np.concatenate([arr for arr in flattened]))

        env_obs = np.stack(env_obs)
        return env_obs
```
the error in this file has disappeared. but another error came out
```
Logging to ./ppo_logs/1744108024/PPO_1
Traceback (most recent call last):
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/train_sb3.py", line 332, in <module>
    main()
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/train_sb3.py", line 268, in main
    model.learn(
  File "/home/panwei/conda/envs/metasim/lib/python3.10/site-packages/stable_baselines3/ppo/ppo.py", line 311, in learn
    return super().learn(
  File "/home/panwei/conda/envs/metasim/lib/python3.10/site-packages/stable_baselines3/common/on_policy_algorithm.py", line 324, in learn
    continue_training = self.collect_rollouts(self.env, callback, self.rollout_buffer, n_rollout_steps=self.n_steps)
  File "/home/panwei/conda/envs/metasim/lib/python3.10/site-packages/stable_baselines3/common/on_policy_algorithm.py", line 218, in collect_rollouts
    new_obs, rewards, dones, infos = env.step(clipped_actions)
  File "/home/panwei/conda/envs/metasim/lib/python3.10/site-packages/stable_baselines3/common/vec_env/base_vec_env.py", line 222, in step
    return self.step_wait()
  File "/home/panwei/conda/envs/metasim/lib/python3.10/site-packages/stable_baselines3/common/vec_env/dummy_vec_env.py", line 59, in step_wait
    obs, self.buf_rews[env_idx], terminated, truncated, self.buf_infos[env_idx] = self.envs[env_idx].step(  # type: ignore[assignment]
  File "/home/panwei/conda/envs/metasim/lib/python3.10/site-packages/stable_baselines3/common/monitor.py", line 94, in step
    observation, reward, terminated, truncated, info = self.env.step(action)
  File "/home/panwei/RoboVerse/roboverse_learn/humanoidbench_rl/sb3_wrapper_mujoco.py", line 72, in step
    metasim_observation, metasim_reward, terminated, truncated, info = self.env.step(action_dict)
  File "/home/panwei/RoboVerse/metasim/sim/env_wrapper.py", line 88, in step
    reward = self._get_reward()
  File "/home/panwei/RoboVerse/metasim/sim/env_wrapper.py", line 115, in _get_reward
    final_reward += reward_func(self.handler.robot.name)(states) * reward_weight
  File "/home/panwei/RoboVerse/metasim/cfg/tasks/humanoidbench/base_cfg.py", line 153, in __call__
    stable_rewards = StableReward(self.robot_name)(states)
  File "/home/panwei/RoboVerse/metasim/cfg/tasks/humanoidbench/base_cfg.py", line 105, in __call__
    head_height(state),
  File "/home/panwei/RoboVerse/metasim/utils/humanoid_robot_util.py", line 23, in head_height
    return envstate[f"{_METASIM_BODY_PREFIX}mid360_link"]["pos"][2]  # TODO: isaaclab don't have head site
KeyError: 'metasim_body_mid360_link'
```

EmptyBlueBox, 2025-04-08:
This issue comes from infrastructure and asset mismatches between the simulators. I'm actively working to resolve them, which may take a few days.

EmptyBlueBox, 2025-04-08:
Fixed in the PR #52

EmptyBlueBox, 2025-04-08:
> This issue comes from infrastructure and asset mismatches between the simulators. I'm actively working to resolve them, which may take a few days.

Actually a few hours.

closed 2025-04-08 by EmptyBlueBox

Fisher-Wang, 2025-04-08:
> > This issue comes from infrastructure and asset mismatches between the simulators. I'm actively working to resolve them, which may take a few days.
> 
> Actually a few hours.

Niubi!

WeisonWEileen, 2025-04-09:
Niubi!