RoboVerseOrg/RoboVerse #51: Release of Assets and Scenes

yzhang4179, 2025-04-08:
Thanks for releasing Roboverse!
I am wondering why can we expect the fully release of all assets, such as assets and scenes in Calvin or ARNOLD?

charlietcheng, 2025-04-08:
Hi Yixin. I'm assuming you're referring to when we will release the dataset. This might take another week or two as we're doing a final inspection with regarding to the licenses for each individual asset. We might release a preview version of the dataset in the meantime (probably within a few days)

Fisher-Wang, 2025-04-13:
Hi, thanks for your interest! We will release the data stage by stage. CALVIN is almost ready and should be released in a couple of days. Please stay tuned!

Please feel free to reopen if you encounter further issues.

closed 2025-04-13 by Fisher-Wang