RoboVerseOrg/RoboVerse #58: docker

Nimingez, 2025-04-09:
when i use docker
98.26   × Failed to build `isaacsim-extscache-physics==4.2.0.2`
98.26   ├─▶ The build backend returned an error
98.26   ╰─▶ Call to `nvidia_stub.buildapi.build_wheel` failed (exit status: 1)
98.26 
98.26       [stdout]
98.26       Downloading wheel
98.26       isaacsim_extscache_physics-4.2.0.2-cp310-none-manylinux_2_34_x86_64.whl
98.26 
98.26       [stderr]
98.26       INFO:nvidia-stub:Testing wheel
98.26       isaacsim_extscache_physics-4.2.0.2-cp310-none-manylinux_2_34_x86_64.whl
98.26       against tag cp310-none-manylinux_2_34_x86_64
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/wheel.py",
98.26       line 177, in download_wheel
98.26           return download_manual(wheel_directory, distribution, version)
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/wheel.py",
98.26       line 158, in download_manual
98.26           file_hash.hexdigest() == sha
98.26       Traceback (most recent call last):
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/wheel.py",
98.26       line 177, in download_wheel
98.26           return download_manual(wheel_directory, distribution, version)
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/wheel.py",
98.26       line 158, in download_manual
98.26           file_hash.hexdigest() == sha
98.26       AssertionError: Downloaded wheel and SHA256 don't match!
98.26       56b32d2abe4015972108c45fbe3675be187aaad9d2c78fb9c6afe6489c816747,
98.26       61f13f6d1b0aeef9ab522a7c3c7ffd414ac390840ae855a80fa49d558990481b
98.26 
98.26       During handling of the above exception, another exception occurred:
98.26 
98.26       Traceback (most recent call last):
98.26         File "<string>", line 11, in <module>
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/buildapi.py",
98.26       line 29, in build_wheel
98.26           return download_wheel(pathlib.Path(wheel_directory),
98.26       config_settings)
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/wheel.py",
98.26       line 179, in download_wheel
98.26           report_install_failure(distribution, version, exception_context)
98.26         File
98.26       "/home/nmz/.cache/uv/builds-v0/.tmpyxZlCv/lib/python3.10/site-packages/nvidia_stub/error.py",
98.26       line 63, in report_install_failure
98.26           raise InstallFailedError(
98.26       nvidia_stub.error.InstallFailedError:
98.26       *******************************************************************************
98.26 
98.26       The installation of isaacsim-extscache-physics for version 4.2.0.2
98.26       failed.
98.26 
98.26       This is a special placeholder package which downloads a real wheel
98.26       package
98.26       from https://pypi.nvidia.com. If https://pypi.nvidia.com is not
98.26       reachable, we
98.26       cannot download the real wheel file to install.
98.26 
98.26       You might try installing this package via
98.26       ```
98.26       $ pip install --extra-index-url https://pypi.nvidia.com
98.26       isaacsim-extscache-physics
98.26       ```
98.26 
98.26       Here is some debug information about your platform to include in any bug
98.26       report:
98.26 
98.26       Python Version: CPython 3.10.16
98.26       Operating System: Linux 5.15.0-136-generic
98.26       CPU Architecture: x86_64
98.26       nvidia-smi command not found. Ensure NVIDIA drivers are installed.
98.26 
98.26       *******************************************************************************
98.26 
98.26 
98.26       hint: This usually indicates a problem with the package or the build
98.26       environment.
98.26   help: `isaacsim-extscache-physics` (v4.2.0.2) was included
98.26         because `roboverse-py[isaaclab]` (v0.1.17) depends on
98.26         `isaacsim-extscache-physics==4.2.0.2`
------

Fisher-Wang, 2025-04-09:
```
98.26 nvidia-smi command not found. Ensure NVIDIA drivers are installed.
```

This line of error message suggest that you didn't install the nvidia driver correctly. Please refer to https://roboverse.wiki/metasim/get_started/prerequisite and ensure nvidia driver 535 or higher is installed.

Nimingez, 2025-04-09:
my pc is ubuntu 20.04  nvidia-smi 550.54.14 nvcc-V 12.4

Fisher-Wang, 2025-04-09:
Could you show the output of nvidia-smi? Since the error message indicate that this command is not found.

Nimingez, 2025-04-09:
[image]

Fisher-Wang, 2025-04-09:
This is weird...

Fisher-Wang, 2025-04-09:
Have you ever tried IsaacLab on your laptop? If not, I suggest install the standalone IsaacLab first following its [official guidance](https://isaac-sim.github.io/IsaacLab/main/source/setup/installation/index.html) and see if there will still be the same error message.

Fisher-Wang, 2025-04-10:
Hi, have you solved this problem?

Fisher-Wang, 2025-04-10:
Please feel free to reopen if you didn't solve it

closed 2025-04-10 by Fisher-Wang