RoboVerseOrg/RoboVerse #63: Missing dataset when trying to run "Replay Trajectories LIBERO"

trannguyenle95, 2025-04-09:
Hi,

I tried to run "python metasim/scripts/replay_demo.py --sim=isaaclab --task=LiberoPickButter" and I got this error says:
`Exception: File roboverse_data/assets/libero/COMMON/stable_hope_objects/butter/usd/butter.usd neither exists in the local directory nor exists in the huggingface dataset. Please report this issue to the developers.`

Also, when I tried to download dataset locally with "git clone git@hf.co:datasets/RoboVerseOrg/roboverse_data" i got this error
`remote: 
remote: ========================================================================
remote: 
remote: ERROR: Repository not found

remote: 
remote: ========================================================================
remote: 
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
`
**Note: I got ssh and Huggingface token set up correctly. This is tested.**

Fisher-Wang, 2025-04-09:
We haven't release the data yet. It will be coming soon. Please stay tuned!

Fisher-Wang, 2025-04-13:
Please feel free to reopen if you encounter further issues.

closed 2025-04-13 by Fisher-Wang