RoboVerseOrg/RoboVerse #64: Use squash & merge when merging PR to main branch

dementrock, 2025-04-10:
Hi all, I suggest you use squash & merge when merging pull requests to the main branch (rather than naive merge). This will help keep the commit history cleaner. You can configure this to be the only allowed merge mechanism to avoid accidental errors.

Fisher-Wang, 2025-04-10:
Thank you, Rocky! This is a valuable suggestion. I also wanted to ask—is it a good practice to directly push to the main branch for quick fixes? Currently some members of the team follow this approach for small changes. Would love to hear your thoughts on this!

dementrock, 2025-04-10:
@Fisher-Wang I recommend doing pull requests for all changes, large and small. This will become more important once you add CI.

geng-haoran, 2025-04-10:
We finished it!

closed 2025-04-10 by geng-haoran