RoboVerseOrg/RoboVerse #69: GPU-based environment stepping without CPU-GPU transfer bottlenecks

younggyoseo, 2025-04-10:
Hi, I'd like to suggest to add a feature that makes environment return GPU tensors instead of returning numpy arrays. I guess the team is already working on the problem but leaving the issue for record.

I found that current code is moving GPU tensors to CPU -- I guess this is compatibility for codebases that assume environments give numpy arrays. But this is causing a serious overhead for each environment step.

For instance, I made the following changes and was able to get 2x speed-up when I'm using 4 parallel HumanoidBench environments:
- Removing GPU-CPU transfer in `humanoid_obs_flatten_func` 
- Re-writing numpy-based `StableReward` and `BaseLocomotionReward` with torch
Maybe all these things can be done in torch and final env wrapper convert tensors to numpy arrays if necessary.

With these, I'm still getting very slow fps with 1024 parallel environments but I guess that's relevant to previous issue about rendering (https://github.com/RoboVerseOrg/RoboVerse/issues/40)

Fisher-Wang, 2025-04-10:
Thank you @younggyoseo ! This is a valuable suggestion. Could @EmptyBlueBox also take a look at this issue?

EmptyBlueBox, 2025-04-14:
Sure, I can fix it these days.

Fisher-Wang, 2025-04-14:
> Sure, I can fix it these days.

I'm doing this on #92. Yutong could you help to review the code and fit the rl codes to it?

EmptyBlueBox, 2025-04-15:
I believe this issue has been largely resolved in #92, #96, and #107. On MuJoCo, I've observed a 100% speed improvement. Feel free to try the latest code and check it out!

closed 2025-04-15 by Fisher-Wang