RoboVerseOrg/RoboVerse #72: quick start：2. Add New Robot some error

zxczcggghkk, 2025-04-11:
run :python get_started/2_add_new_robot.py  --sim mujoco --headless
error:Traceback (most recent call last):
  File "/home/zxc/RoboVerse/get_started/2_add_new_robot.py", line 233, in <module>
    obs, reward, success, time_out, extras = env.step(actions)
  File "/home/zxc/RoboVerse/metasim/sim/env_wrapper.py", line 89, in step
    states = self.handler.get_states()
  File "/home/zxc/RoboVerse/metasim/sim/mujoco/mujoco.py", line 279, in get_states
    state[object_type][obj.name]["dof_pos_target"] = {
  File "/home/zxc/RoboVerse/metasim/sim/mujoco/mujoco.py", line 280, in <dictcomp>
    joint_name: self.actions_cache[0]["dof_pos_target"][joint_name] for joint_name in joint_names
KeyError: ''

run:python get_started/2_add_new_robot.py  --sim genesis

There is no h1 model in the generated video

[image]

geng-haoran, 2025-04-12:
I can reproduce your error! It's being fixed. It seems to be a dynamic problem for articulated objects with Genesis

geng-haoran, 2025-04-12:
If you change fix_base_link to True at line 62, everything works fine. I think the reason for the error is a dynamic processing or urdf loading problem in genesis. It's not a problem with infra. I'm still trying to figure it out.

geng-haoran, 2025-04-12:
[image]

zxczcggghkk, 2025-04-14:
> If you change fix_base_link to True at line 62, everything works fine. I think the reason for the error is a dynamic processing or urdf loading problem in genesis. It's not a problem with infra. I'm still trying to figure it out.如果你在第 62 行将 fix_base_link 更改为 True，一切正常。我认为错误是由于 Genesis 中的动态处理或 URDF 加载问题造成的。这与 Infra 无关。我仍在努力弄清楚。

thank you

closed 2025-11-25 by Fisher-Wang