RoboVerseOrg/RoboVerse #77: How to use extension of ROS 2 Humble with Isaac Sim on Ubuntu 20.04?

David-Kingsman, 2025-04-12:
I am currently trying to integrate Isaac Sim with ROS 2 Humble on Ubuntu 20.04 (I understand that Humble officially targets Ubuntu 22.04). I would like to ask:
Is there any recommended way to run Isaac Sim + ROS 2 Humble on Ubuntu 20.04?
Would you suggest building from source or using containerized solutions?
Are there any known compatibility issues I should be aware of (e.g. with rclcpp, ament, colcon, etc.)?

Fisher-Wang, 2025-04-13:
Hi! We haven't fit our framework to ROS, though we plan to do so. For now, you can refer to the [Isaac Sim's official guidance](https://docs.isaacsim.omniverse.nvidia.com/latest/ros2_tutorials/ros2_landing_page.html).

Fisher-Wang, 2025-04-13:
Please feel free to reopen if you have further issues!

closed 2025-04-13 by Fisher-Wang