RoboVerseOrg/RoboVerse #93: Failed to run some simulator in docker container

Poorunga, 2025-04-14:
I ran the metasim container and tried to run an example, but encountered several problems, as follows:

## Q1: The metasim environment cannot run isaaclab, no module erorr. ``
mamba activate metasim
```
(metasim) root@df1a608bcdc6:/home/root/RoboVerse# python get_started/0_static_scene.py  --sim isaaclab
```
log:
```
Traceback (most recent call last):
  File "/home/root/RoboVerse/metasim/sim/isaaclab/isaaclab.py", line 19, in <module>
    from omni.isaac.lab.app import AppLauncher
ModuleNotFoundError: No module named 'omni.isaac.lab'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/root/RoboVerse/get_started/0_static_scene.py", line 102, in <module>
    env_class = get_sim_env_class(SimType(args.sim))
  File "/home/root/RoboVerse/metasim/utils/setup_util.py", line 126, in get_sim_env_class
    raise e
  File "/home/root/RoboVerse/metasim/utils/setup_util.py", line 121, in get_sim_env_class
    from metasim.sim.isaaclab import IsaaclabEnv
  File "/home/root/RoboVerse/metasim/sim/isaaclab/__init__.py", line 1, in <module>
    from .isaaclab import IsaaclabEnv, IsaaclabHandler  # noqa: F401
  File "/home/root/RoboVerse/metasim/sim/isaaclab/isaaclab.py", line 21, in <module>
    from isaaclab.app import AppLauncher
ModuleNotFoundError: No module named 'isaaclab'
```

## Q2: The metasim environment cannot run sapine, and the sapine version is incorrect.
```
(metasim) root@df1a608bcdc6:/home/root/RoboVerse# python get_started/0_static_scene.py  --sim sapien
```
log:
```
/home/root/conda/envs/metasim/lib/python3.10/site-packages/sapien/_vulkan_tricks.py:37: UserWarning: Failed to find Vulkan ICD file. This is probably due to an incorrect or partial installation of the NVIDIA driver. SAPIEN will attempt to provide an ICD file anyway but it may not work.
  warn(
Traceback (most recent call last):
  File "/home/root/RoboVerse/get_started/0_static_scene.py", line 103, in <module>
    env = env_class(scenario)
  File "/home/root/RoboVerse/metasim/sim/env_wrapper.py", line 67, in __init__
    self.handler = cls(*args, **kwargs)
  File "/home/root/RoboVerse/metasim/sim/parallel.py", line 111, in __new__
    return base_cls(scenario)
  File "/home/root/RoboVerse/metasim/sim/sapien/sapien2.py", line 44, in __init__
    assert parse_version(sapien.__version__) < parse_version("3.0.0a0"), "Sapien version should be lower than 3.0.0"
AssertionError: Sapien version should be lower than 3.0.0
```

## Q3: The metasim isaac gym environment cannot run isaacgym, prompting that so is missing
mamba activate metasim_isaacgym
```
(metasim) root@df1a608bcdc6:/home/root/RoboVerse# python get_started/0_static_scene.py  --sim isaacgym
```
log:
```
Importing module 'gym_38' (/home/root/conda/envs/metasim_isaacgym/lib/python3.8/site-packages/isaacgym/_bindings/linux-x86_64/gym_38.so)
                    ERROR    IsaacGym is not installed, please install it first                                                                                                                   setup_util.py:133
Traceback (most recent call last):
  File "get_started/0_static_scene.py", line 102, in <module>
    env_class = get_sim_env_class(SimType(args.sim))
  File "/home/root/RoboVerse/metasim/utils/setup_util.py", line 134, in get_sim_env_class
    raise e
  File "/home/root/RoboVerse/metasim/utils/setup_util.py", line 129, in get_sim_env_class
    from metasim.sim.isaacgym.isaacgym import IsaacgymEnv
  File "/home/root/RoboVerse/metasim/sim/isaacgym/__init__.py", line 3, in <module>
    from .isaacgym import IsaacgymEnv, IsaacgymHandler
  File "/home/root/RoboVerse/metasim/sim/isaacgym/isaacgym.py", line 7, in <module>
    from isaacgym import gymapi, gymtorch, gymutil  # noqa: F401
  File "/home/root/conda/envs/metasim_isaacgym/lib/python3.8/site-packages/isaacgym/__init__.py", line 5, in <module>
    from isaacgym import gymapi
  File "/home/root/conda/envs/metasim_isaacgym/lib/python3.8/site-packages/isaacgym/gymapi.py", line 104, in <module>
    _import_active_version()
  File "/home/root/conda/envs/metasim_isaacgym/lib/python3.8/site-packages/isaacgym/gymapi.py", line 63, in _import_active_version
    module = importlib.import_module(package_path)
  File "/home/root/conda/envs/metasim_isaacgym/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libpython3.8.so.1.0: cannot open shared object file: No such file or directory
```

Poorunga, 2025-04-14:
more info:
```
uname -a
Linux wjz-v100 6.5.0-28-generic #29~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Apr  4 14:39:20 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
```
```
nvidia-smi 
Mon Apr 14 12:20:50 2025       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.216.03             Driver Version: 535.216.03   CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  Tesla V100-SXM2-16GB           On  | 00000000:21:01.0 Off |                    0 |
| N/A   29C    P0              38W / 300W |    351MiB / 16384MiB |      1%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      2569      C   /usr/NX/bin/nxnode.bin                      347MiB |
+---------------------------------------------------------------------------------------+
```

Fisher-Wang, 2025-04-14:
Hi!

## Q1

Could you check `pip list | grep isaaclab` and see if isaaclab exists in your env?

## Q2

Could you check `pip list | grep sapien` and see if sapien exists in your env?

## Q3

Please check the tips in [doc](https://roboverse.wiki/metasim/get_started/installation#install-isaacgym)

Fisher-Wang, 2025-04-14:
I just noticed your username in docker container is root. So are you building docker as root account? I'm not sure if this could be the problem but maybe there will be some permission issues

Fisher-Wang, 2025-04-14:
Thanks for your feedback. Q1 is also strange to me. Maybe you can try `docker compose up  --build &> logs.txt` and paste the build log in `log.txt`. Q2 I just realize we are using sapien3 in docker, so running with `--sim=sapien3` should work well.

Poorunga, 2025-04-14:
Q1: I will re-build the image.
Q2:
> Q2 I just realize we are using sapien3 in docker, so running with --sim=sapien3 should work well.
It works. Thank you.

Poorunga, 2025-04-14:
```
(metasim) root@df1a608bcdc6:/home/root/RoboVerse/third_party# pip list | grep isaac
isaacsim                   4.2.0.2
isaacsim-app               4.2.0.2
isaacsim-asset             4.2.0.2
isaacsim-benchmark         4.2.0.2
isaacsim-code-editor       4.2.0.2
isaacsim-core              4.2.0.2
isaacsim-cortex            4.2.0.2
isaacsim-example           4.2.0.2
isaacsim-extscache-kit     4.2.0.2
isaacsim-extscache-kit-sdk 4.2.0.2
isaacsim-extscache-physics 4.2.0.2
isaacsim-gui               4.2.0.2
isaacsim-kernel            4.2.0.2
isaacsim-replicator        4.2.0.2
isaacsim-rl                4.2.0.2
isaacsim-robot             4.2.0.2
isaacsim-robot-motion      4.2.0.2
isaacsim-robot-setup       4.2.0.2
isaacsim-ros1              4.2.0.2
isaacsim-ros2              4.2.0.2
isaacsim-sensor            4.2.0.2
isaacsim-storage           4.2.0.2
isaacsim-template          4.2.0.2
isaacsim-test              4.2.0.2
isaacsim-utils             4.2.0.2
omni-isaac-lab_assets      0.1.4          /home/root/packages/IsaacLab/source/extensions/omni.isaac.lab_assets
omni-isaac-lab_tasks       0.10.18        /home/root/packages/IsaacLab/source/extensions/omni.isaac.lab_tasks
```
Although there is no isaaclab, there are a series of isaac-related packages. These are installed during docker build. @Fisher-Wang

Fisher-Wang, 2025-04-14:
Hi, have you resolved the issue? If not, I recommend sharing the build log so I can assist you further.

Poorunga, 2025-04-15:
> Hi, have you resolved the issue? If not, I recommend sharing the build log so I can assist you further.

I recompiled on another machine with an RTX 4090 (the network speed is too slow, still compiling). I will give you feedback when I have more information. Thank you.

Fisher-Wang, 2025-04-15:
You are welcome!

chenyenru, 2025-04-15:
Hey @Poorunga I ran into Q2 too and resolved it by specifying `sapien3` instead. It seems like sapien refers to sapien 2 and sapien3 to 3 because it's under development. 

The command looks like:

```shell
python get_started/0_static_scene.py  --sim sapien3
```

Poorunga, 2025-04-16:
Another question：
```
Exception: File roboverse_data/robots/franka/mjcf/panda.xml neither exists in the local directory nor exists in the huggingface dataset. Please report this issue to the developers.
```

Fisher-Wang, 2025-05-07:
Hi! Sorry for the late reply. Didn't solve the issue?

chenyenru, 2025-05-07:
Q2: Seems like the data is wip, see https://github.com/RoboVerseOrg/RoboVerse/issues/14
Q3: Assuming you're running the command to build curobo from source, you need to either downgrade your CUDA toolkit version to 11.8 or reinstall your PyTorch with the same version that is compiled for 12.4. 

https://discuss.pytorch.org/t/install-pytorch-with-cuda-12-1/174294/10

I think the torch version will depend on simulators you're using. Check the version, go to the PyTorch installation website, and find the corresponding installation command (then run `uv pip install ...`)
https://pytorch.org/get-started/previous-versions/#:~:text=pip%20install%20torch%3D%3D2.4.0%20torchvision%3D%3D0.19.0%20torchaudio%3D%3D2.4.0%20%2D%2Dindex%2Durl%20https%3A//download.pytorch.org/whl/cu124

Fisher-Wang, 2025-08-29:
> Q2: Seems like the data is wip, see [#14](https://github.com/RoboVerseOrg/RoboVerse/issues/14) Q3: Assuming you're running the command to build curobo from source, you need to either downgrade your [...]

Thanks for your feedback! We are working on to replace curobo with Pyroki, which could solve the installation issue

closed 2025-08-29 by Fisher-Wang