RoboVerseOrg/RoboVerse #115: How to use the data obtained from mimicgen for domain randomization in Roboverse

HanwenWangBUPT, 2025-04-16:
Hi, thank you for your outstanding work. Could you please explain how to use the data obtained from mimicgen for domain randomization in Roboverse to achieve better generalization performance on the Roboverse benchmark?"

wkwan7, 2025-04-16:
Hi, for now I recommend directly using the Isaac Lab Mimic [data generation functions](https://isaac-sim.github.io/IsaacLab/main/source/overview/teleop_imitation.html#generating-additional-demonstrations) provided i official IsaacLab, as it is convenient and consistent with the implementation used in MimicGen and the relevant experiments in this paper.

HanwenWangBUPT, 2025-04-16:
Thank you for your reply. I have two more questions:

1.During data augmentation, are trajectory augmentation (mimicgen) and domain randomization (e.g., lighting, camera pose) applied simultaneously in the simulator?

2.Which factor plays a key role in sim-to-real transfer?

wkwan7, 2025-04-16:
For point 1, in the experiments presented in our paper, we investigated each component separately to validate their individual effectiveness. However, combining them is certainly feasible as well.

For point 2, I believe all of these factors are important for sim-to-real transfer. The relative importance of each one depends on your specific simulation/real-world setup and algorithm—for example, the number of source demonstrations and the perception sensitivity of your model.

closed 2025-04-16 by wkwan7