RoboVerseOrg/RoboVerse #117: collect demo error

zxczcggghkk, 2025-04-16:
run:python metasim/scripts/collect_demo.py --task=CloseBox --num_envs=1 --run_all --sim=mujoco
ERROR    KeyError: 'metasim_body_panda_hand'   

[image]

Fisher-Wang, 2025-04-16:
Hi! This error message doesn't appear on my side. Could you update your local repo by `git pull` and retry it?

Fisher-Wang, 2025-04-17:
Please feel free to reopen if you encounter further issues

closed 2025-04-17 by Fisher-Wang

Poorunga, 2025-04-21:
+1.
 DEBUG    Joint box_joint of object box_base has position [1.78]                                                                                                                  checkers.py:68
[04/21/25 10:19:53] ERROR    KeyError: 'metasim_body_panda_hand'                                                                                                                                close_box_cfg.py:38
                    DEBUG    Joint box_joint of object box_base has position [1.74]                                                                                                                  checkers.py:68
                    ERROR    KeyError: 'metasim_body_panda_hand'