RoboVerseOrg/RoboVerse #118: roboverse_learn:eval.py

zxczcggghkk, 2025-04-16:
run:python eval.py --sim mujoco --task CloseBox --algo ACT

Traceback (most recent call last):
  File "/home/zxc/RoboVerse/roboverse_learn/eval.py", line 215, in <module>
    main()
  File "/home/zxc/RoboVerse/roboverse_learn/eval.py", line 152, in main
    "rgb": torch.stack([env["cameras"]["camera0"]["rgb"] for env in obs]),
TypeError: 'TensorState' object is not iterable

Fisher-Wang, 2025-04-17:
Hi! This bug should be fixed for now, could you update the repo with `git pull` and try again?

zxczcggghkk, 2025-04-18:
Thanks for your work, and when I run the script again.
error：Traceback (most recent call last):
  File "/home/zxc/RoboVerse/roboverse_learn/eval.py", line 210, in <module>
    main()
  File "/home/zxc/RoboVerse/roboverse_learn/eval.py", line 172, in main
    obs, reward, success, time_out, extras = env.step(action)
  File "/home/zxc/RoboVerse/metasim/sim/env_wrapper.py", line 103, in step
    self.handler.set_dof_targets(self.handler.robot.name, actions)
  File "/home/zxc/RoboVerse/metasim/sim/mujoco/mujoco.py", line 376, in set_dof_targets
    actuator.ctrl = target_pos
  File "/home/zxc/miniconda3/envs/py310/lib/python3.10/site-packages/torch/_tensor.py", line 1196, in __array__
    return self.numpy().astype(dtype, copy=False)
TypeError: can't convert cuda:0 device type tensor to numpy. Use Tensor.cpu() to copy the tensor to host memory first.

### I tried adding the following code to line 159 of eval.py and was able to run it successfully

[image]