RoboVerseOrg/RoboVerse #121: Real2sim and AI-Generate Tasks

gaolin131, 2025-04-16:
The content for Real2sim and AI-Generate Tasks was not found in the code and wiki.

louhz, 2025-04-17:
Hi there, thanks for your interest!

Our Real2Sim toolset will be released soon. We’re currently working on pip package management to make installation and updates smoother. This release will include:

An advanced structure-from-motion pipeline

Fast Gaussian and mesh reconstruction workflows

Instructions on how to build URDF (MJCF) from the reconstructed scene

Stay tuned for updates! If you have any questions, feel free to ask.

Peihao2021, 2025-04-17:
Hi, thanks for your interest in AI-Generate Task! The script, documentation and data will be released soon. 

Our pipeline would provide an example inference script for users to retrieve from the customized assets list, and arrange them automatically in the format required by MetaSim. 

Then, you could directly visualize the inital scene and modify it based on customized needs!

Please stay tuned!

ziyanGao, 2025-06-13:
Thank you for your incredible work on RoboVerse! Could you share an update on whether the Real2sim released timeline is still on track?