RoboVerseOrg/RoboVerse #123: Can't find roboverse_learn/eval_vla.py

4evertutelary, 2025-04-17:
Dear Contributor:
Thanks a lot for your great work!

When I follow https://roboverse.wiki/roboverse_learn/imitation_learning/openvla, I got 2 issues:
1 can't access https://github.com/wbjsamuel/RoboVerse-OpenVLA , url from "Clone the [RoboVerse] version of OpenVLA and install the environment as required."
2 can't find code roboverse_learn/eval_vla.py

Please help to check when you have free time.

Best Regards
Sean

woltium, 2025-04-17:
same question

wbjsamuel, 2025-04-19:
Hi, thanks for pointing it out! We haven't updated the OpenVLA-related content in the wiki. You can directly use the [official OpenVLA repo](https://github.com/openvla/openvla) for finetuning. And [roboverse_learn/eval.py](https://github.com/RoboVerseOrg/RoboVerse/blob/main/roboverse_learn/eval.py) also supports VLA evaluation now.

closed 2025-04-19 by wbjsamuel

4evertutelary, 2025-04-19:
@wbjsamuel 
Hi, where to download openvla checkpoint? Is it ok to download from https://huggingface.co/openvla/openvla-7b-finetuned-libero-object/tree/main ?

Which task can I try?  CloseBox?

wbjsamuel, 2025-04-19:
We use https://huggingface.co/openvla/openvla-7b for per-task finetuning during our experiments. You can fine-tune your own version based on the data you collected or your own data recipe.

4evertutelary, 2025-04-19:
> We use https://huggingface.co/openvla/openvla-7b for per-task finetuning during our experiments. You can fine-tune your own version based on the data you collected or your own data recipe.

Run:
python roboverse_learn/eval.py --robot franka --task LiberoPickMilk --algo openvla --checkpoint_path ~/code/RoboVerse1/openvla-7b-finetuned-libero-object/

Get Err:
ERROR    diffuse_texture path /home/zhaoweiping/code/RoboVerse1/roboverse_data/assets/libero/COMMON/stable_hope_objects/milk/usd/../textures/texture_map.png (parsed from usd_util.py:86
                             ../textures/texture_map.png) does not exist, please fix manually as relative path!

Any suggestion?

shapera-lab, 2025-09-03:
Run LiberoPickAlphabetSoup with mujoco:
python roboverse_learn/eval.py --task LiberoPickAlphabetSoup --sim mujoco --algo openvla --checkpoint_path /root/VLA_Dataset/models/

Got Err:
Traceback (most recent call last):
  File "/mnt/bn/isp-vladata-lq2/zhangshaopeng/VLA_Algo/RoboVerse/roboverse_learn/eval.py", line 197, in <module>
    main()
  File "/mnt/bn/isp-vladata-lq2/zhangshaopeng/VLA_Algo/RoboVerse/roboverse_learn/eval.py", line 108, in main
    policyRunner: PolicyRunner = runnerCls(
  File "/mnt/bn/isp-vladata-lq2/zhangshaopeng/VLA_Algo/RoboVerse/roboverse_learn/algorithms/base_runner.py", line 35, in __init__
    self._init_policy(**kwargs)
  File "/mnt/bn/isp-vladata-lq2/zhangshaopeng/VLA_Algo/RoboVerse/roboverse_learn/algorithms/vla_runner.py", line 28, in _init_policy
    self.DATA_STAT = json.load(open(self.dataset_info_path))
FileNotFoundError: [Errno 2] No such file or directory: '/root/VLA_Dataset/models/openvla-7b/dataset_statistics.json'
Segmentation fault (core dumped)