RoboVerseOrg/RoboVerse #128: Bug happened when opening multiple cameras

geng-haoran, 2025-04-18:
There are some bugs when opening multiple cameras:
For example:
scenario.cameras = [
    PinholeCameraCfg(width=1024, height=1024, pos=(0.0, 0.0, 1.5), look_at=(1.0, 0.0, 0.0)),
    PinholeCameraCfg(width=1024, height=1024, pos=(1.5, 0.0, 1.5), look_at=(0.0, 0.0, 0.0)),
]
It seems the returned rgb and depth are all from the second camera.

songlin, 2025-04-21:
I will look into this and get back to you ASAP.

martin-sedlacek, 2025-04-28:
Hi, I will piggyback on this issue while it is open - just tried adding another camera to the RT demo (6_advanced_rendering.py) like this:
```
scenario.cameras = [
    PinholeCameraCfg(width=1024, height=1024, pos=(1.5, -1.5, 1.5), look_at=(0.5, 0.0, 0.0), name="cam0"),
    PinholeCameraCfg(width=1024, height=1024, pos=(2.5, -1.5, 2), look_at=(0.0, 0.0, 0.0), name="cam1")
]
```

But it seems like isaaclab is not creating them properly:
```
Traceback (most recent call last):
  File "/home/ros/RoboVerse/get_started/test_multi_cam_raytracing.py", line 108, in <module>
    env = env_class(scenario)
  File "/home/ros/RoboVerse/metasim/sim/env_wrapper.py", line 37, in __init__
    self.handler.launch()
  File "/home/ros/RoboVerse/metasim/sim/isaaclab/isaaclab.py", line 87, in launch
    self.env: EmptyEnv = gym.make("MetaSimEmptyTaskEnv", cfg=env_cfg)
  File "/home/ros/conda/envs/metasim/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 742, in make
    env = env_creator(**env_spec_kwargs)
  File "/home/ros/RoboVerse/metasim/sim/isaaclab/empty_env.py", line 56, in __init__
    super().__init__(cfg, render_mode, **kwargs)
  File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 122, in __init__
    self._setup_scene()
  File "/home/ros/RoboVerse/metasim/sim/isaaclab/empty_env.py", line 59, in _setup_scene
    self.func["_setup_scene"](self)
  File "/home/ros/RoboVerse/metasim/sim/isaaclab/env_overwriter.py", line 333, in _setup_scene
    add_cameras(env, self.cameras)
  File "/home/ros/RoboVerse/metasim/sim/isaaclab/isaaclab_helper.py", line 302, in add_cameras
    _add_pinhole_camera(env, camera)
  File "/home/ros/RoboVerse/metasim/sim/isaaclab/isaaclab_helper.py", line 282, in _add_pinhole_camera
    env.scene.sensors[camera.name] = TiledCamera(
  File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/tiled_camera.py", line 93, in __init__
    super().__init__(cfg)
  File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 130, in __init__
    self.cfg.spawn.func(
  File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/utils.py", line 251, in wrapper
    prim = func(prim_paths[0], cfg, *args, **kwargs)
  File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/sensors/sensors.py", line 87, in spawn_camera
    raise ValueError(f"A prim already exists at path: '{prim_path}'.")
ValueError: A prim already exists at path: '/World/envs/env_0/camera0'.
```
@geng-haoran seems like you did not run into this and can at least instantiate an environment with multiple cameras out of the box?

geng-haoran, 2025-04-29:
Looking into it now!

Fisher-Wang, 2025-04-29:
> Hi, I will piggyback on this issue while it is open - just tried adding another camera to the RT demo (6_advanced_rendering.py) like this:
> 
> ```
> scenario.cameras = [
>     PinholeCameraCfg(width=1024, height=1024, pos=(1.5, -1.5, 1.5), look_at=(0.5, 0.0, 0.0), name="cam0"),
>     PinholeCameraCfg(width=1024, height=1024, pos=(2.5, -1.5, 2), look_at=(0.0, 0.0, 0.0), name="cam1")
> ]
> ```
> 
> But it seems like isaaclab is not creating them properly:
> 
> ```
> Traceback (most recent call last):
>   File "/home/ros/RoboVerse/get_started/test_multi_cam_raytracing.py", line 108, in <module>
>     env = env_class(scenario)
>   File "/home/ros/RoboVerse/metasim/sim/env_wrapper.py", line 37, in __init__
>     self.handler.launch()
>   File "/home/ros/RoboVerse/metasim/sim/isaaclab/isaaclab.py", line 87, in launch
>     self.env: EmptyEnv = gym.make("MetaSimEmptyTaskEnv", cfg=env_cfg)
>   File "/home/ros/conda/envs/metasim/lib/python3.10/site-packages/gymnasium/envs/registration.py", line 742, in make
>     env = env_creator(**env_spec_kwargs)
>   File "/home/ros/RoboVerse/metasim/sim/isaaclab/empty_env.py", line 56, in __init__
>     super().__init__(cfg, render_mode, **kwargs)
>   File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/envs/direct_rl_env.py", line 122, in __init__
>     self._setup_scene()
>   File "/home/ros/RoboVerse/metasim/sim/isaaclab/empty_env.py", line 59, in _setup_scene
>     self.func["_setup_scene"](self)
>   File "/home/ros/RoboVerse/metasim/sim/isaaclab/env_overwriter.py", line 333, in _setup_scene
>     add_cameras(env, self.cameras)
>   File "/home/ros/RoboVerse/metasim/sim/isaaclab/isaaclab_helper.py", line 302, in add_cameras
>     _add_pinhole_camera(env, camera)
>   File "/home/ros/RoboVerse/metasim/sim/isaaclab/isaaclab_helper.py", line 282, in _add_pinhole_camera
>     env.scene.sensors[camera.name] = TiledCamera(
>   File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/tiled_camera.py", line 93, in __init__
>     super().__init__(cfg)
>   File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sensors/camera/camera.py", line 130, in __init__
>     self.cfg.spawn.func(
>   File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/utils.py", line 251, in wrapper
>     prim = func(prim_paths[0], cfg, *args, **kwargs)
>   File "/home/ros/packages/IsaacLab/source/extensions/omni.isaac.lab/omni/isaac/lab/sim/spawners/sensors/sensors.py", line 87, in spawn_camera
>     raise ValueError(f"A prim already exists at path: '{prim_path}'.")
> ValueError: A prim already exists at path: '/World/envs/env_0/camera0'.
> ```
> 
> [@geng-haoran](https://github.com/geng-haoran) seems like you did not run into this and can at least instantiate an environment with multiple cameras out of the box?
> 
> I'm running this inside the default docker container on a local machine.

Hi, I follow your modification and change `6_advanced_rendering.py` into
```
"""This script is used to test the static scene."""

from __future__ import annotations

from typing import Literal

try:
    import isaacgym  # noqa: F401
except ImportError:
    pass

import os

import imageio
import numpy as np
import rootutils
import torch
import tyro
from loguru import logger as log
from rich.logging import RichHandler
[... 117 lines cut ...]
                    "panda_finger_joint1": 0.04,
                    "panda_finger_joint2": 0.04,
                },
            },
        },
    }
]
obs, extras = env.reset(states=init_states)
os.makedirs("get_started/output", exist_ok=True)
save_path = f"get_started/output/6_advanced_rendering_{args.sim}_{args.render.mode}.png"
log.info(f"Saving image to {save_path}")
img_cat = np.concatenate([obs.cameras["cam0"].rgb[0].cpu().numpy(), obs.cameras["cam1"].rgb[0].cpu().numpy()], axis=1)
imageio.imwrite(save_path, img_cat)
```

and run
```
python get_started/6_advanced_rendering.py --render.mode=pathtracing
```

the rendered image `get_started/output/6_advanced_rendering_isaaclab_pathtracing.png` is

[image]

Everything works fine. Could you please check whether the local branch is up-to-date with the remote one?

Fisher-Wang, 2025-04-29:
Also an example script (`get_started/multiple_cameras.py`) is added for multiple cameras. @geng-haoran The key is to explicitly name different cameras when there are multiple cameras

martin-sedlacek, 2025-04-29:
Seems like my codebase was indeed not up to date :) works now, thanks for the check!

closed 2025-04-29 by Fisher-Wang