RoboVerseOrg/RoboVerse #130: xxxx neither exists in the local directory nor exists in the huggingface dataset

Poorunga, 2025-04-19:
```
python metasim/scripts/replay_demo.py --sim=isaaclab --task=StackCube --num_envs=4 --robot=iiwa
[04/19/25 05:57:35] INFO     Args: Args(task='StackCube', robot='iiwa', scene=None, render=RenderCfg(mode='raytracing'), random=RandomizationCfg(camera=False, light=False, ground=False,         replay_demo.py:72
                             reflection=False, table=False, wall=False, scene=False, level=0), sim='isaaclab', renderer=None, num_envs=4, try_add_table=True, object_states=False, split='all',                    
                             headless=False, save_image_dir='tmp', save_video_path=None, stop_on_runout=False)                                                                                                     
Traceback (most recent call last):
  File "/home/root/RoboVerse/metasim/scripts/replay_demo.py", line 258, in <module>
    main()
  File "/home/root/RoboVerse/metasim/scripts/replay_demo.py", line 151, in main
    scenario = ScenarioCfg(
  File "<string>", line 21, in __init__
  File "/home/root/RoboVerse/metasim/utils/configclass.py", line 415, in _combined
    f1(*args, **kwargs)
  File "/home/root/RoboVerse/metasim/cfg/scenario.py", line 99, in __post_init__
    check_asset(self.robot, self.sim)
  File "/home/root/RoboVerse/metasim/cfg/scenario.py", line 33, in check_asset
    check_and_download(obj.usd_path)
  File "/home/root/RoboVerse/metasim/utils/hf_util.py", line 24, in check_and_download
    check_and_download_single(filepath)
  File "/home/root/RoboVerse/metasim/utils/hf_util.py", line 56, in check_and_download_single
    raise Exception(
Exception: File data_isaaclab/robots/iiwa/iiwa.usd neither exists in the local directory nor exists in the huggingface dataset. Please report this issue to the developers.
```

There is no problem with the network connection.

dujl, 2025-04-26:
I encountered the same problem

Fisher-Wang, 2025-04-30:
Hi, thanks for you interest! We are still organizing the data for release and will do it stage by stage. For iiwa robot, I will check and release it, hopefully tomorrow. Please stay tuned!

jmcoholich, 2025-05-12:
I am also interested in the full dataset release

Poorunga, 2025-05-19:
I am also interested in the full dataset release