RoboVerseOrg/RoboVerse #147: get_started：4_motion_planing

whx123123, 2025-04-29:
I'm working on replacing the Franka model with another robotic arm model in the get_started 04_motion_planing project. I have managed to locate the relevant USD and URDF files for the robotic arm in the example-assets. Additionally, I have created and registered the corresponding cfg files in the metsiom/cfg/robots directory, and I have also written the relevant robot YAML files in the configs/robot section of curobo. However, the program still fails to run. I hope to get the correct replacement steps.

geng-haoran, 2025-04-30:
Could you please share the error message you're seeing, along with more details about the arm setup and the configuration (cfg) you're using? With a bit more context, we would be easier to help you debug the issue!

whx123123, 2025-05-06:
Let me first explain my thought process.
1.Based on "get_started/02_add_new_robot", I learned how to replace the system robot. However, in "4_motion_planning", I switched to using the CFG format for configuration. I created a custom CFG in /metasim/cfg/robots and registered it in the __init__ file. My CFG-related files are as follows:
@configclass
class PiperCfg(BaseRobotCfg):
    """Cfg for the Franka Emika Panda robot.

    Args:
        BaseRobotCfg (_type_): _description_
    """

    name: str = "piper"
    num_joints: int = 8
    fix_base_link: bool = True
    usd_path="/home/whx/RoboVerse/get_started/example_assets/piper/piper_1.usd"
    # mjcf_path: str = "roboverse_data/robots/franka/mjcf/panda.xml"
    # urdf_path: str = "roboverse_data/robots/franka/urdf/panda.urdf"  # work for pybullet and sapien
    # urdf_path: str = "roboverse_data/robots/franka/urdf/franka_panda.urdf"  # work for isaacgym
    enabled_gravity: bool = False
    enabled_self_collisions: bool = False
    actuators: dict[str, BaseActuatorCfg] = {
        "joint1": BaseActuatorCfg(velocity_limit=2.175),
        "joint2": BaseActuatorCfg(velocity_limit=2.175),
        "joint4": BaseActuatorCfg(velocity_limit=2.175),
        "joint3": BaseActuatorCfg(velocity_limit=2.175),
        "joint5": BaseActuatorCfg(velocity_limit=2.61),
        "joint6_to_gripper_base": BaseActuatorCfg(velocity_limit=2.61),
        "joint7": BaseActuatorCfg(velocity_limit=0.2, is_ee=True),
        "joint8": BaseActuatorCfg(velocity_limit=0.2, is_ee=True),
    }
    joint_limits: dict[str, tuple[float, float]] = {
        "joint1": (-2.8973, 2.8973),
        "joint2": (-1.7628, 1.7628),
        "joint3": (-2.8973, 2.8973),
        "joint4": (-3.0718, -0.0698),
        "joint5": (-2.8973, 2.8973),
        "joint6_to_gripper_base": (-0.0175, 3.7525),
        "joint7": (0.0, 0.04),  # 0.0 close, 0.04 open
        "joint8": (0.0, 0.04),  # 0.0 close, 0.04 open
    }
    ee_body_name: str = "gripper_link"

    default_joint_positions: dict[str, float] = {
        "joint1": 0.0,
        "joint2": 0.0,
        "joint3": 0.0,
        "joint4": 0.0,
        "joint5": 0.0,
        "joint6_to_gripper_base": 0.0,
        "joint7": 0.0,
        "joint8": 0.0,
    }

    # TODO: Make it more elegant
    gripper_release_q = [0.04, 0.04]
    gripper_actuate_q = [0.0, 0.0]

    curobo_ref_cfg_name: str = "piper.yml"
    curobo_tcp_rel_pos: tuple[float, float, float] = [0.0, 0.0, 0.10312]
    curobo_tcp_rel_rot: tuple[float, float, float] = [0.0, 0.0, 0.0]
2.Then, while writing the CFG, I noticed that a YAML file was required. Following the instructions, I made the necessary configurations in both the assets and configs directories under /third_party/curobo/src/curobo/content. In the configs directory, I created a YAML file as well as a corresponding YAML file under spheres, as shown below:
robot.yml:
robot_cfg:
  kinematics:
    use_usd_kinematics: False
    isaac_usd_path: ""
    usd_path: "/home/whx/RoboVerse/third_party/curobo/src/curobo/content/assets/robot/piper/piper_1.usd"
    usd_robot_root: "/piper"
    usd_flip_joints: ["joint1","joint2","joint3","joint4", "joint5","joint6_to_gripper_base","joint7","joint8"]
    usd_flip_joints: {
    "joint1": "Z",
[... 95 lines cut ...]
    robot=robot,
    try_add_table=False,
    sim=args.sim,
    headless=args.headless,
    num_envs=args.num_envs,
)
        "robots": {
            "piper": {
                "pos": torch.tensor([0.0, 0.0, 0.3]),
                "rot": torch.tensor([1.0, 0.0, 0.0, 0.0]),
                "dof_pos": {
                    "joint1": 0.0,
                    "joint2": 0.0,
                    "joint3": 0.0,
                    "joint4": 0.0,
                    "joint5": 0.0,
                    "joint6_to_gripper_base": 0.0,
                    "joint7": 0.0,
                    "joint8": 0.0,
                },
            },
        },
    }
    for _ in range(args.num_envs)
]

whx123123, 2025-05-06:
I would like to ask whether my approach is correct, or if it's possible to provide a sample example, so that those of us who are passionate about Roboverse can better understand how to use it for new scientific research.

Fisher-Wang, 2025-05-07:
Hi! I suggest  first test whether the robot cfg is valid by running
```
python metasim/scripts/random_action_really_pure.py --robot=piper --sim=isaaclab
```

If the simulator could run without error, and the robot doesn't break apart, then the robot cfg is valid.

You can refer to franka as expected behavior:
```
python metasim/scripts/random_action_really_pure.py --robot=franka
```

closed 2025-11-25 by Fisher-Wang