RoboVerseOrg/RoboVerse #161: how to use rlbench in headless server in CogACT(vla)

Luo-Zhongwei, 2025-05-05:


geng-haoran, 2025-05-05:
Could you please explain more about CogACT(vla)? Are you running this model as a baseline and would like to test in our rlbench tasks?

Luo-Zhongwei, 2025-05-06:
> Could you please explain more about CogACT(vla)? Are you running this model as a baseline and would like to test in our rlbench tasks?

yes yes ,I want to running this model as a baseline , test in robotverse about rlbench 

here is related doc

https://github.com/microsoft/CogACT

CogACT: A Foundational Vision-Language-Action Model for Synergizing Cognition and Action in Robotic Manipulation

@article{li2024cogact,
  title={CogACT: A Foundational Vision-Language-Action Model for Synergizing Cognition and Action in Robotic Manipulation},
  author={Li, Qixiu and Liang, Yaobo and Wang, Zeyu and Luo, Lin and Chen, Xi and Liao, Mozheng and Wei, Fangyun and Deng, Yu and Xu, Sicheng and Zhang, Yizhong and others},
  journal={arXiv preprint arXiv:2411.19650},
  year={2024}
}

Luo-Zhongwei, 2025-05-13:
hi，