RoboVerseOrg/RoboVerse #170: Building Docker image

JaehongJaehongMin, 2025-05-08:
Thank you for sharing your great work!

I'v encountered an issue when I build up the docker file.

I used this command in the tutorials.
`docker compose up --build -d && docker exec -it metasim bash`

 => ERROR [metasim  7/23] RUN mamba init bash                                                                                                     0.1s
------                                                                                                                                                 
 > [metasim  7/23] RUN mamba init bash:                                                                                                                
0.111 The following arguments were not expected: bash init
0.111 Run with --help for more information.
------
failed to solve: process "/bin/bash -c mamba init bash" did not complete successfully: exit code: 109

JaehongJaehongMin, 2025-05-08:
I've tried `Run mamba init`, but didn't work.

geng-haoran, 2025-05-08:
Thank you for your interest and the issue. @Fisher-Wang Could you please have a look at this?

Fisher-Wang, 2025-05-08:
Hi, this seems due to the recent upgrade of miniforge3 which remove the `mamba init` subcommand. Changing Line 47 of `Dockerfile` into `RUN conda init bash` should be working well.

JaehongJaehongMin, 2025-05-08:
Thank you for your suggestion, @Fisher-Wang ! 
I think we can use `mamba shell init bash` too.

JaehongJaehongMin, 2025-05-08:
Thank you @Fisher-Wang !

I've tried `mamba shell init -s bash`, but it makes another error. And your solution works well!

Fisher-Wang, 2025-05-08:
You're welcome!

PR #171 (2025-05-08): [fix] resolve mamba init command not found in docker
Fixed #170
closed 2025-05-08 by Fisher-Wang