RoboVerseOrg/RoboVerse #172: Building Docker container

JaehongJaehongMin, 2025-05-08:
Sorry, I encountered another issue while building the Docker container.
After I changed `mamba init bash` to `conda init`, the container ran normally,
but the following error showed up.

```

'mamba' is running as a subprocess and can't modify the parent shell.
Thus you must initialize your shell before using activate and deactivate.

To initialize the current bash shell, run:
    $ eval "$(mamba shell hook --shell bash)"
and then activate or deactivate with:
    $ mamba activate
To automatically initialize all future (bash) shells, run:
    $ mamba shell init --shell bash --root-prefix=~/.local/share/mamba
If your shell was already initialized, reinitialize your shell with:
    $ mamba shell reinit --shell bash
Otherwise, this may be an issue. In the meantime you can run commands. See:
    $ mamba run --help

Supported shells are {bash, zsh, csh, posix, xonsh, cmd.exe, powershell, fish, nu}.
critical libmamba Shell not initialized
Remember to run: xhost +local:docker on the host to enable GUI applications.

```

Fisher-Wang, 2025-05-08:
Hi, I think this message doesn't affect using docker. If it does or if you want to suppress it, please update the repo with `git pull`, it should be working well now!

JaehongJaehongMin, 2025-05-08:
After I pull from the origin, it works like a charm.
Sorry for the confusing and thank you again for your works!

JaehongJaehongMin, 2025-05-08:
Well.. I encountered the segmentation fault when I access to bashrc file again.
But I guess I found the reason.

Commenting out this line , "export LD_LIBRARY_PATH=...", in bashrc file solved the problem.

Fisher-Wang, 2025-05-08:
Hi! Thank you for pointing this out. This is a bug we didn't realize. The latest repo should have fixed this. You can update and have a try!

closed 2025-05-08 by Fisher-Wang

JaehongJaehongMin, 2025-05-09:
Thank you so much for your quick response. The container is now running perfectly!

PR #174 (2025-05-08): [fix] fix docker library conflict
This PR fix docker library conflict by updating LD_LIBRARY_PATH only when metasim_isaacgym conda env is activated

This PR also add section comments for `Dockerfile` for better readability

Fixed #172