RoboVerseOrg/RoboVerse #173: get_started/0_static_scene.py

JaehongJaehongMin, 2025-05-08:
I'm trying these exmaples to check out all of the simulations run well.

[image]

Mujoco, Pybullet, sapien3 are fine.
But isaaclab, gym, genesis raise errors.

- Genesis
```
Traceback (most recent call last):
  File "/workspace/RoboVerse/get_started/0_static_scene.py", line 143, in <module>
    obs, extras = env.reset(states=init_states)
  File "/workspace/RoboVerse/metasim/sim/env_wrapper.py", line 92, in reset
    self.handler.set_states(states, env_ids=env_ids)
  File "/workspace/RoboVerse/metasim/sim/genesis/genesis.py", line 171, in set_states
    obj_inst.set_pos(np.array([states_flat[env_id][obj.name]["pos"] for env_id in env_ids]))
ValueError: only one element tensors can be converted to Python scalars
```

- Isaacgym
```
PyTorch version 2.4.1+cu118
Device count 1
/home/user/conda/envs/metasim_isaacgym/lib/python3.8/site-packages/isaacgym/_bindings/src/gymtorch
Using /home/user/.cache/torch_extensions/py38_cu118 as PyTorch extensions root...
Emitting ninja build file /home/user/.cache/torch_extensions/py38_cu118/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Segmentation fault (core dumped)
```

- Isaaclab
```
Traceback (most recent call last):
  File "get_started/0_static_scene.py", line 100, in <module>
    env_class = get_sim_env_class(SimType(args.sim))
  File "/workspace/RoboVerse/metasim/utils/setup_util.py", line 121, in get_sim_env_class
    from metasim.sim.isaaclab import IsaaclabEnv
  File "/workspace/RoboVerse/metasim/sim/isaaclab/__init__.py", line 1, in <module>
    from .isaaclab import IsaaclabEnv, IsaaclabHandler  # noqa: F401
  File "/workspace/RoboVerse/metasim/sim/isaaclab/isaaclab.py", line 17, in <module>
    from .env_overwriter import IsaaclabEnvOverwriter
  File "/workspace/RoboVerse/metasim/sim/isaaclab/env_overwriter.py", line 7, in <module>
    from .isaaclab_helper import add_cameras, add_lights, add_objects, add_robot, add_sensors, get_pose
  File "/workspace/RoboVerse/metasim/sim/isaaclab/isaaclab_helper.py", line 137, in <module>
    def add_objects(env: "EmptyEnv", objects: list[BaseObjCfg]) -> None:
TypeError: 'type' object is not subscriptable
```

geng-haoran, 2025-05-08:
For isaaclab, are you using python <= 3.8?

geng-haoran, 2025-05-08:
I will check genesis and isaacgym. For IsaacGym, this can happen on different systems, and it seems that there are some bugs in isaacgym itself.

JaehongJaehongMin, 2025-05-08:
> For isaaclab, are you using python <= 3.8?

I used `isaaclab` mamba environment, created by the Dockerfile, and it's python version is 3.8.20.

> I will check genesis and isaacgym.

Thank you so much for your hardwork!

JaehongJaehongMin, 2025-05-08:
I ran the Docker container from scratch, and Sapien showed an error. I believe I had fixed this issue before, but I don't remember exactly how. Apologies for the inconvenience.

```
Traceback (most recent call last):
  File "/home/user/workspace/RoboVerse/get_started/0_static_scene.py", line 101, in <module>
    env = env_class(scenario)
  File "/home/user/workspace/RoboVerse/metasim/sim/env_wrapper.py", line 83, in __init__
    self.handler.launch()
  File "/home/user/workspace/RoboVerse/metasim/sim/sapien/sapien2.py", line 365, in launch
    self._build_sapien()
  File "/home/user/workspace/RoboVerse/metasim/sim/sapien/sapien2.py", line 53, in _build_sapien
    self.renderer = sapien_core.SapienRenderer()  # Create a renderer
RuntimeError: createInstance: No Vulkan extensions found for window surface creation (hint: set VK_ICD_FILENAMES to `locate icd.json`).

```

Fisher-Wang, 2025-05-08:
For isaacgym, please use the `metasim_isaacgym` conda env in the docker.

For other simulator backends (including isaaclab and genesis), please use the `metasim` conda env in the docker.

For viewer issues, did you run `xhost +local:docker` as instructed in [doc](https://roboverse.wiki/metasim/get_started/docker)?

JaehongJaehongMin, 2025-05-09:
> For isaacgym, please use the metasim_isaacgym conda env in the docker.

I think we're saying the same environment `metasim_isaacgym`.

> For other simulator backends (including isaaclab and genesis), please use the metasim conda env in the docker.

I used `metasim_isaacgym` for isaaclab and issacgym, and `metasim` for the others.

> For viewer issues, did you run xhost +local:docker as instructed in [doc](https://roboverse.wiki/metasim/get_started/docker)?

Yes, I've already chekced that xclock works fine and pops up windows in docker container.

Isaacgym is still showing segmentation fault and isaaclab Type error.
I've tried genesis with and without `--headless` option.
It coudn't find display without `--headless`

```
[Genesis] [01:22:50] [ERROR] No display detected. Use `show_viewer=False` for headless mode.
```

and it raised EGL Error with `--headless`.
```
libEGL warning: failed to open /dev/dri/renderD128: Permission denied

libEGL warning: failed to open /dev/dri/card1: Permission denied

OpenGL.raw.EGL._errors.EGLError: EGLError(
        err = EGL_NOT_INITIALIZED,
        baseOperation = eglInitialize,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x74f7d0ef9fc0>,
                c_long(0),
                c_long(0),
        ),
        result = 0
)

```

Fisher-Wang, 2025-05-09:
> I used `metasim_isaacgym` for isaaclab and issacgym, and `metasim` for the others.

For isaaclab, please do use the `metasim` env instead of `metasim_isaacgym`

For isaacgym, could you provide more information so that we can help you debug:
- For the host machine, what OS are you using? what's the nvidia driver version and cuda toolkit version?
- In the docker, could you try to reproduce the segfault and show the last lines of system log? `sudo dmesg | last -50` should be fine.

For genesis, OS and driver information could also be helpful. Meanwhile, could you update the repo to include the modification by #176 and try again to see if the problem still exists?

raiderss, 2025-05-09:
Amazing work! As someone from Eyes Store developing FiveM resources, I find this really inspiring.

JaehongJaehongMin, 2025-05-09:
> For isaaclab, please do use the metasim env instead of metasim_isaacgym

I apologize for the confusion.

I ran isaaclab with `metasim` environment, and I stuck at OmniGraphSettings [Isaaclab](https://github.com/isaac-sim/IsaacLab/issues/987).

> For the host machine, what OS are you using? what's the nvidia driver version and cuda toolkit version?

I'm currently using `Ubuntu 22.04`, `nvidia driver 550.144.03` and `nvcc 12.4.131` in my host machine.

> Meanwhile, could you update the repo to include the modification by https://github.com/RoboVerseOrg/RoboVerse/pull/176 and try again to see if the problem still exists?

Now genesis runs without `--headless` option! Thank you very much. But it still raises EGL error with `--headless` option

> In the docker, could you try to reproduce the segfault and show the last lines of system log? sudo dmesg | tail -50 should be fine.

```
[05/09/25 02:38:55] INFO     Using simulator: isaacgym                                                       0_static_scene.py:99
PyTorch version 2.4.1+cu118
Device count 1
/home/user/conda/envs/metasim_isaacgym/lib/python3.8/site-packages/isaacgym/_bindings/src/gymtorch
Using /home/user/.cache/torch_extensions/py38_cu118 as PyTorch extensions root...
Emitting ninja build file /home/user/.cache/torch_extensions/py38_cu118/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Segmentation fault (core dumped)
(metasim_isaacgym) user@min:~/workspace/RoboVerse$ sudo dmesg | tail -50
[600867.504326] audit: type=1400 audit(1746756224.812:3634): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600867.504331] audit: type=1400 audit(1746756224.812:3635): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600911.547805] kauditd_printk_skb: 14 callbacks suppressed
[600911.547812] audit: type=1400 audit(1746756268.857:3650): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600911.547819] audit: type=1400 audit(1746756268.857:3651): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600911.564327] audit: type=1400 audit(1746756268.873:3652): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600911.564332] audit: type=1400 audit(1746756268.873:3653): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600911.579391] audit: type=1400 audit(1746756268.888:3654): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[600911.579398] audit: type=1400 audit(1746756268.888:3655): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[601116.054901] audit: type=1400 audit(1746756473.369:3656): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[601116.054908] audit: type=1400 audit(1746756473.369:3657): apparmor="DENIED" operation="open" class="file" profile="/usr/bin/evince" name="/var/lib/aspell/en-common.rws" pid=3397033 comm="evince" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[601116.655611] audit: type=1400 audit(1746756473.970:3658): apparmor="DENIED" operation="connect" class="file" profile="/usr/bin/evince" name="/run/user/1000/gvfsd/socket-rZ8QCRfT" pid=3397033 comm="evince" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[601116.656209] audit: type=1400 audit(1746756473.970:3659): apparmor="DENIED" operation="connect" class="file" profile="/usr/bin/evince" name="/run/user/1000/gvfsd/socket-at5bSnG5" pid=3397033 comm="evince" requested_mask="wr" denied_mask="wr" fsuid=1000 ouid=1000
[601116.742492] audit: type=1107 audit(1746756474.057:3660): pid=786 uid=102 auid=4294967295 ses=4294967295 subj=unconfined msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/org/freedesktop/hostname1" interface="org.freedesktop.DBus.Properties" member="GetAll" mask="send" name=":1.9413" pid=3612 label="snap.firefox.firefox" peer_pid=2563156 peer_label="unconfined"
                 exe="/usr/bin/dbus-daemon" sauid=102 hostname=? addr=? terminal=?'
[... cut ...] sk="r" denied_mask="r" fsuid=1000 ouid=0
[602036.627376] audit: type=1400 audit(1746757393.961:3729): apparmor="DENIED" operation="open" class="file" profile="snap-update-ns.firefox" name="/proc/2675458/maps" pid=2675458 comm="5" requested_mask="r" denied_mask="r" fsuid=1000 ouid=0
[602978.410011] pt_main_thread[2791321]: segfault at 0 ip 0000000000000000 sp 00007ffe3b6c0ad8 error 14 likely on CPU 3 (core 3, socket 0)
[602978.410019] Code: Unable to access opcode bytes at 0xffffffffffffffd6.
```

Fisher-Wang, 2025-05-09:
> I ran isaaclab with `metasim` environment, and I stuck at OmniGraphSettings [Isaaclab](https://github.com/isaac-sim/IsaacLab/issues/987).

I see, I encountered this error before, did you try [the solution](https://github.com/isaac-sim/IsaacLab/issues/987#issuecomment-2351033220) in that issue?

Fisher-Wang, 2025-05-09:
For isaacgym, it seems a very tricky problem that I don't know why. Could you try installing isaacgym on your host machine seperately (not within metasim) following its [official guide](https://docs.robotsfan.com/isaacgym/install.html) and see if [this](https://docs.robotsfan.com/isaacgym/install.html#simple-example) will succeed?

JaehongJaehongMin, 2025-05-09:
> I see, I encountered this error before, did you try https://github.com/isaac-sim/IsaacLab/issues/987#issuecomment-2351033220 in that issue?

No, I didn't. I'm runing other codes on my local machine, and I want to use ordinary driver on my computer, not server one.
Are you runing code on your server, and using server driver?

> For isaacgym, it seems a very tricky problem that I don't know why. Could you try installing isaacgym on your host machine seperately (not within metasim) following its [official [...]

Okay! I'll try it.

Fisher-Wang, 2025-05-19:
Hi! Just a quick follow-up -- did you solve the issue?

easylab2007, 2025-07-08:
yes it is work well by installing isaacgym on my host machine seperately    and   only use the container of Roboverse have this issue.

and all those 4 simulation is work well with my Roboverse container:
python get_started/0_static_scene.py  --sim mujoco --headless
python get_started/0_static_scene.py  --sim sapien3 --headless
python get_started/0_static_scene.py  --sim pybullet --headless
python get_started/0_static_scene.py  --sim genesis --headless

only this two simulation is not work wtih my Roboverse container:
python get_started/0_static_scene.py  --sim isaacgym --headless
python get_started/0_static_scene.py  --sim isaaclab --headless

for isaacgym  have this issue:
Segmentation fault (core dumped)

for isaaclab have this issue:
File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/isaacsim/extscache/omni.usd-1.12.2+10a4b5c0.lx64.r.cp310/omni/usd/_impl/__init__.py", line 37, in <module>
    from .._usd import *
ImportError: [libomni.usd.so](http://libomni.usd.so/): cannot open shared object file: No such file or directory

Fisher-Wang, 2025-07-09:
For isaaclab, same problem happened on my side too. It seems NVIDIA recently update their isaacsim4.2 wheel which uses incorrect usd library version. I'll update the docker to use the latest isaaclab2 (isaacsim4.5) instead.

For isaacgym, it works on my side. Could you share the following information to help further debug:
- nvidia driver version (check with nvidia-smi)
- cuda toolkit version on your host machine (check with `nvcc --version` on your host machine)
- system log inside docker (check with `sudo dmesg -T` inside docker)

Fisher-Wang, 2025-07-09:
Update for isaaclab error: We've located the error to be introduced by #274 , not NVIDIA's. We're actively working on this on #333. For now, you can switch to a previous commit (e.g. `2fd9615`) and eveything should be fine with isaaclab