RoboVerseOrg/RoboVerse #184: rootutils and open3d

fengjungui, 2025-05-12:
When I installed,I get some erro:
there is no package "rootutils and open3d",otherwise When I run example of mujoco.——AttributeError: 'resolution' is not a valid attribute for <camera>

fengjungui, 2025-05-12:
[05/12/25 17:48:57] INFO     Using simulator: mujoco                                                                                                                 1_control_robot.py:99
[05/12/25 17:48:58] WARNING  Warning: hard coding decimation to 25 for replaying trajectories                                                                                 mujoco.py:56
Traceback (most recent call last):
  File "/home/roboter/文档/FengLearning/MyRoboverse/RoboVerse/get_started/1_control_robot.py", line 101, in <module>
    env = env_class(scenario)
  File "/home/roboter/文档/FengLearning/MyRoboverse/RoboVerse/metasim/sim/env_wrapper.py", line 83, in __init__
    self.handler.launch()
  File "/home/roboter/文档/FengLearning/MyRoboverse/RoboVerse/metasim/sim/mujoco/mujoco.py", line 60, in launch
    model = self._init_mujoco()
  File "/home/roboter/文档/FengLearning/MyRoboverse/RoboVerse/metasim/sim/mujoco/mujoco.py", line 128, in _init_mujoco
    mjcf_model.worldbody.add("camera", name=f"{camera.name}_custom", **camera_params)
  File "/home/roboter/.local/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 598, in add
    return self.insert(element_name, position=None, **kwargs)
  File "/home/roboter/.local/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 623, in insert
    new_element = _make_element(child_spec, self, attributes=kwargs)
  File "/home/roboter/.local/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 96, in _make_element
    return _ElementImpl(spec, parent, attributes)
  File "/home/roboter/.local/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 153, in __init__
    self._check_valid_attribute(attribute_name)
  File "/home/roboter/.local/lib/python3.10/site-packages/dm_control/mjcf/element.py", line 502, in _check_valid_attribute
    raise AttributeError(
AttributeError: 'resolution' is not a valid attribute for <camera>

Fisher-Wang, 2025-05-12:
Hi! For now, could you install the both package and try again?
```
pip install rootutils open3d
```
We will manage the dependencies in a more smooth way!

fengjungui, 2025-05-13:
Yes,it's work now

 

 
&mdash;

closed 2025-05-13 by Fisher-Wang