RoboVerseOrg/RoboVerse #267: [Bug Report] Fix the geneesis simulator error caused by different igl versions in docker

Junb0Dong, 2025-06-14:
When I used docker to run tutorials, I encountered this error when running  `python get_started/0_static_scene.py  --sim genesis`

> Traceback (most recent call last):
  File "/home/RoboVerse/get_started/0_static_scene.py", line 99, in <module>
    env = env_class(scenario)
  File "/home/RoboVerse/metasim/sim/env_wrapper.py", line 84, in __init__
    self.handler.launch()
  File "/home/RoboVerse/metasim/sim/genesis/genesis.py", line 42, in launch
    self.scene_inst.add_entity(gs.morphs.Plane())
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/scene.py", line 359, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/simulator.py", line 134, in _add_entity
    entity = self.rigid_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/solvers/rigid/rigid_solver_decomp.py", line 64, in add_entity
    entity = entity_class(
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 68, in __init__
    self._load_model()
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 87, in _load_model
    self._load_primitive(self._morph, self._surface)
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 189, in _load_primitive
    link._add_geom(
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_link.py", line 173, in _add_geom
    geom = RigidGeom(
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 94, in __init__
    self._preprocess()
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 147, in _preprocess
    sdf_closest_vert = self._compute_closest_verts(query_points)
  File "/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py", line 229, in _compute_closest_verts
    _, closest_faces, _ = igl.signed_distance(query_points, self._init_verts, self._init_faces)
ValueError: too many values to unpack (expected 3)

I found that this is caused by inconsistent versions of the igl library. The new version of igl will return multiple values(>3 values).
So we can modify the code to solve the error in the file `/home/user/conda/envs/metasim/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_geom.py` 223 line.
```python
    def _compute_sd(self, query_points):
        result = igl.signed_distance(query_points, self._sdf_verts, self._sdf_faces)
        sd = result[0]
        return sd

    def _compute_closest_verts(self, query_points):
        _, closest_faces, _, _ = igl.signed_distance(query_points, self._init_verts, self._init_faces)
        verts_ids = self._init_faces[closest_faces]
        verts_ids = verts_ids[
            np.arange(len(query_points)).astype(int),
            np.argmin(np.linalg.norm(self._init_verts[verts_ids] - query_points[:, None, :], axis=-1), axis=-1),
        ]
        return verts_ids
```

Fisher-Wang, 2025-06-15:
Hi, thanks for your interest! Then it should be a bug of Genesis instead of RoboVerse?

Fisher-Wang, 2025-06-15:
Please feel free to raise it in [genesis repo](https://github.com/Genesis-Embodied-AI/Genesis)

closed 2025-06-15 by Fisher-Wang