RoboVerseOrg/RoboVerse #329: [Question] HumanoidBench assets

humphreymunn, 2025-07-08:
Hi, thank you for creating this repo, I am curious when HumanoidBench assets will be integrated into the repo? At the moment it is not possible to do most of the HumanoidBench tasks. I am trying to use HumanoidBench with isaaclab. Thank you!

geng-haoran, 2025-07-09:
Thank you! @MurphyZhao04 @WeisonWEileen Could you please check this issue?

Priokin, 2025-07-09:
Thanks for the heads-up—we are finishing the `roboverse_data` release that bundles all HumanoidBench meshes and XMLs, so Isaac Gym, MuJoCo, and MJX can already load the tasks once the data lands, while Isaac Lab are still with bugs and should be resolved within about a month; for now, the tasks that run reliably are **stand, walk, run, sit, stair, slide, and crawl**

closed 2025-07-09 by humphreymunn