RoboVerseOrg/RoboVerse #343: [Question]collect_demo.py fails on CALVIN-A: missing calvin_table_A.urdf & question about get_task returning class instead of instance

whatcansay, 2025-07-11:
Hi maintainers,
First of all, thank you for the impressive roboverse project—its design and reproducibility have been extremely helpful for our research!
While running
```bash
python metasim/scripts/collect_demo.py --task=MoveSliderLeftA --sim=pybullet --run-all --headless
```
I ran into two issues that I could not resolve after checking the docs and existing issues. Any guidance would be greatly appreciated.

1. `get_task` returns a class instead of an instance
In `metasim.utils.setup_util`, the helper
```python
task = get_task(args.task)
```
returns the class object itself, not an instance.
Consequently, attempting to access task_cls.sim_params raises an AttributeError; only after explicitly calling task_cls() can I proceed.
Is this behavior intentional (to allow users to override constructor arguments), or should get_task return an already-instantiated task?
If the current design is on purpose, what is the recommended way to retrieve sim_params before instantiation?

2. Missing URDF file for CALVIN table A
When I continue to the CALVIN-A environment, the script aborts with
`Exception: File roboverse_data/assets/calvin/COMMON/table_a/urdf/calvin_table_A.urdf neither exists in the local directory nor exists in the Hugging Face dataset. Please report this issue to the developers.`
The download completed without errors, CALVIN assets entirely missing from Hugging Face dataset – collect_demo.py raises missing URDF error for any CALVIN task
Could you advise how to obtain the missing URDF file? Should I manually download it from an alternative source, or will the dataset be updated?
Thanks for your time and for open-sourcing this great work!

whatcansay, 2025-07-12:
Hi RoboVerse team,
    Thanks for your response.
    I have read through the “Dataset and Benchmark → Task Descriptions” section, but it is still hard to tell which individual tasks are fully working right now—i.e., which ones have:
        1.runnable code in the public GitHub repo, and 
        2.ready-to-use dataset / configuration files on Hugging Face.
    Could you please either:
        - publish a concise matrix (benchmark → task name → code status → HF dataset link), or
        - point me to the exact file / page that keeps this information up-to-date?
    Knowing the precise list will save a lot of trial-and-error—thank you very much!

whatcansay, 2025-07-12:
I also encountered the same problem when trying tasks in Libero
```
Exception: File roboverse_data/assets/libero/COMMON/stable_hope_objects/butter/mjcf/butter.xml neither exists in the local directory nor exists in the huggingface dataset. Please report this issue to the developers
```

geng-haoran, 2025-07-13:
Please check this for a preview. These are tasks that guarantee can work: https://roboverse.wiki/dataset_benchmark/benchmark/results

geng-haoran, 2025-07-13:
We are working on the full release