RoboVerseOrg/RoboVerse #358: [Bug Report] Errors with get_started/5_hybrid_sim.py

song-siqi, 2025-07-15:
### Describe the bug

I tried `python get_started/5_hybrid_sim.py  --sim mujoco --renderer isaaclab` with the default environment on IsaacLab1.4 and met with errors, which looks like two bugs: joint position out of range and camera settings.

### Steps to reproduce

I initialized the environment following the documentation and then

```
> python get_started/5_hybrid_sim.py  --sim isaaclab --renderer mujoco
[07/16/25 02:01:58] INFO     Args: Args(robot='franka', sim='isaaclab',            5_hybrid_sim.py:50
                             renderer='mujoco', num_envs=1, headless=False)                          
                    INFO     File roboverse_data/robots/franka/usd/franka_v2.usd found  hf_util.py:36
                             in local directory.                                                     
                    INFO     Using simulator: isaaclab, renderer: mujoco          5_hybrid_sim.py:105
                    WARNING  Warning: hard coding decimation to 25 for replaying         mujoco.py:48
                             trajectories                                                            
[WARN][AppLauncher]: There are no arguments attached to the ArgumentParser object. If you have your own arguments, please load your own arguments before calling the `AppLauncher.add_app_launcher_args` method. This allows the method to check the validity of the arguments and perform checks for argument names.
[INFO][AppLauncher]: Loading experience file: /home/song-siqi/Projects/RoboVerse/third_party/IsaacLab/source/apps/isaaclab.python.rendering.kit
[Warning] [omni.isaac.kit.simulation_app] Modules: ['omni.kit_app'] were loaded before SimulationApp was started and might not be loaded correctly.
[Warning] [omni.isaac.kit.simulation_app] Please check to make sure no extra omniverse or pxr modules are imported before the call to SimulationApp(...)
Loading user config located at: '/home/song-siqi/miniconda3/envs/robo-310-mujoco-isaaclab14/lib/python3.10/site-packages/omni/data/Kit/Isaac-Sim/4.2/user.config.json'
[Info] [carb] Logging to file: /home/song-siqi/miniconda3/envs/robo-310-mujoco-isaaclab14/lib/python3.10/site-packages/omni/logs/Kit/Isaac-Sim/4.2/kit_20250716_020200.log
2025-07-15 18:02:00 [0ms] [Warning] [omni.kit.app.plugin] No crash reporter present, dumps uploading isn't available.
DEBUG:asyncio:Using selector: EpollSelector

|---------------------------------------------------------------------------------------------|
| Driver Version: 535.247.01    | Graphics API: Vulkan
|=============================================================================================|
| GPU | Name                             | Active | LDA | GPU Memory | Vendor-ID | LUID       |
|     |                                  |        |     |            | Device-ID | UUID       |
|     |                                  |        |     |            | Bus-ID    |            |
|---------------------------------------------------------------------------------------------|
| 0   | NVIDIA GeForce RTX 4060 Ti       | Yes: 0 |     | 8434    MB | 10de      | 0          |
|     |                                  |        |     |            | 2803      | f596b70e.. |
|     |                                  |        |     |            | 8         |            |
|=============================================================================================|
| OS: 22.04.5 LTS (Jammy Jellyfish) ubuntu, Version: 22.04.5, Kernel: 6.8.0-60-generic
| XServer Vendor: The X.Org Foundation, XServer Version: 12101004 (1.21.1.4)
| Processor: AMD Ryzen 5 5600 6-Core Processor | Cores: 6 | Logical: 12
|---------------------------------------------------------------------------------------------|
| Total Memory (MB): 15869 | Free Memory: 7197
| Total Page/Swap (MB): 3904 | Free Page/Swap: 2872
|---------------------------------------------------------------------------------------------|
2025-07-15 18:02:00 [704ms] [Warning] [gpu.foundation.plugin] IOMMU is enabled.
INFO:omni.kit.telemetry.impl.sentry_extension:sentry is disabled for external build
INFO:omni.kit.telemetry.impl.sentry_extension:sentry is disabled for external build
DEBUG:AutoNode:Defining data type 'any' as 'Any'
DEBUG:AutoNode:Defining data type 'bool' as 'Bool' and array 'BoolArray
DEBUG:AutoNode:Defining data type 'bundle' as 'Bundle'
DEBUG:AutoNode:Defining data type 'colord[3]' as 'Color3d' and array 'Color3dArray
DEBUG:AutoNode:Defining data type 'colorf[3]' as 'Color3f' and array 'Color3fArray
DEBUG:AutoNode:Defining data type 'colorh[3]' as 'Color3h' and array 'Color3hArray
DEBUG:AutoNode:Defining data type 'colord[4]' as 'Color4d' and array 'Color4dArray
DEBUG:AutoNode:Defining data type 'colorf[4]' as 'Color4f' and array 'Color4fArray
DEBUG:AutoNode:Defining data type 'colorh[4]' as 'Color4h' and array 'Color4hArray
DEBUG:AutoNode:Defining data type 'double' as 'Double' and array 'DoubleArray
DEBUG:AutoNode:Defining data type 'double[2]' as 'Double2' and array 'Double2Array
DEBUG:AutoNode:Defining data type 'double[3]' as 'Double3' and array 'Double3Array
DEBUG:AutoNode:Defining data type 'double[4]' as 'Double4' and array 'Double4Array
DEBUG:AutoNode:Defining data type 'execution' as 'Execution'
DEBUG:AutoNode:Defining data type 'float' as 'Float' and array 'FloatArray
DEBUG:AutoNode:Defining data type 'float[2]' as 'Float2' and array 'Float2Array
DEBUG:AutoNode:Defining data type 'float[3]' as 'Float3' and array 'Float3Array
DEBUG:AutoNode:Defining data type 'float[4]' as 'Float4' and array 'Float4Array
DEBUG:AutoNode:Defining data type 'frame[4]' as 'Frame' and array 'FrameArray
DEBUG:AutoNode:Defining data type 'half' as 'Half' and array 'HalfArray
DEBUG:AutoNode:Defining data type 'half[2]' as 'Half2' and array 'Half2Array
DEBUG:AutoNode:Defining data type 'half[3]' as 'Half3' and array 'Half3Array
DEBUG:AutoNode:Defining data type 'half[4]' as 'Half4' and array 'Half4Array
[... 347 lines cut ...]
  <global offwidth="my_width"/>
</visual>
[1]    38997 killed     python get_started/5_hybrid_sim.py --sim mujoco --renderer isaaclab

```

### System Info

Describe the characteristic of your environment:

 - Commit: main branch [commit 3129c39]
 - Simulator Backend Version: Isaac Lab v1.4.1
 - OS: Ubuntu 22.04
 - GPU: RTX 4060ti
 - CUDA: 11.8
 - GPU Driver: 535.247.01

### Additional context

Add any other context about the problem here.

### Checklist

- [ ] I have checked that there is no similar issue in the repo (**required**)
- [ ] I have checked that the issue is not in the simulator backends (IsaacLab, IsaacGym, MuJoCo, etc.) and is related to the RoboVerse repo

qq775193759, 2025-07-16:
I meet the same Errors when I run 1_control_robot.py using MuJoCo with headless mode.

Fisher-Wang, 2025-08-30:
Thanks for reporting this! This bug was fixed in #338

closed 2025-08-30 by Fisher-Wang