RoboVerseOrg/RoboVerse #437: [Bug Report] unable to run 0_static_scene.py on genesis

blyu413, 2025-08-20:
### Describe the bug

I encountered this bug when running `get_started/0_static_scene.py`. I also tried to replay Maniskill tasks using `metasim/scripts/collect_demo.py` on genesis, and I got similar errors.

### Steps to reproduce

Run command:
```
python ./get_started/0_static_scene.py --sim=genesis
```
Error message:
```
Traceback (most recent call last):
  File "/home/blyu/Code/my_verse/./get_started/0_static_scene.py", line 141, in <module>
    obs, extras = env.reset(states=init_states)
  File "/home/blyu/Code/my_verse/metasim/sim/env_wrapper.py", line 96, in reset
    self.handler.set_states(states, env_ids=env_ids)
  File "/home/blyu/Code/my_verse/metasim/sim/base.py", line 88, in set_states
    self._set_states(states, env_ids=env_ids)
  File "/home/blyu/Code/my_verse/metasim/sim/genesis/genesis.py", line 207, in _set_states
    np.array([
  File "/home/blyu/Code/my_verse/metasim/sim/genesis/genesis.py", line 210, in <listcomp>
    for jn in self.get_joint_names(obj.name, sort=False)
  File "/home/blyu/Code/my_verse/metasim/sim/genesis/genesis.py", line 313, in get_joint_names
    joint_names = [
  File "/home/blyu/Code/my_verse/metasim/sim/genesis/genesis.py", line 316, in <listcomp>
    if j.dof_idx_local is not None and j.name != self.object_inst_dict[obj_name].base_joint.name
  File "/home/blyu/Code/my_verse/.venv/lib/python3.10/site-packages/taichi/lang/kernel_impl.py", line 1236, in _getattr
    return x(self)
  File "/home/blyu/Code/my_verse/.venv/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 2854, in base_joint
    return self._joints[0][0]
  File "/home/blyu/Code/my_verse/.venv/lib/python3.10/site-packages/genesis/datatypes.py", line 15, in __getitem__
    result = super().__getitem__(item)
IndexError: list index out of range
```
The `base_joint` in genesis package returns `self._joint[0][0]`, while `self._joint[0]` is an empty list.

### System Info

- Commit: 33efcd3
- Simulator Backend Version: genesis-world 0.3.1
- OS: Ubuntu 22.04
- GPU: RTX 5070Ti laptop
- CUDA: 12.8
- GPU Driver: 570.172.08

### Potential fix

By removing the criterion relates to `base_joint` at [`metasim/sim/genesis/genesis.py` line 317](https://github.com/RoboVerseOrg/RoboVerse/blob/33efcd3dbbbc09d573c00a5518197faa3682b91b/metasim/sim/genesis/genesis.py#L317), I am able to run `./get_started/0_static_scene.py` and replay Maniskill tasks using `metasim/scripts/collect_demo.py`. After this modification, the model can be loaded correctly. I am not very sure if this modification is safe. My modification is given as: 
```
joint_names = [
    j.name
    for j in joints
    f j.dof_idx_local is not None
]
```
I could also check the existence of `base_joint` in advanced, and I got the same simulation result.

Regarding the motion of robot, I noticed some difference on different simulators after modifying the code. I replay the identical `pick_cube` trajectory on mujoco and genesis seperately using the following commands:
```
python metasim/scripts/collect_demo.py --sim=genesis --task=PickCube --num_envs=1 --max_demo_idx=1 --run_all
python metasim/scripts/collect_demo.py --sim=mujoco --task=PickCube --num_envs=1 --max_demo_idx=1 --run_all
```
On genesis:
[image]

On mujoco:
[image]

In these two videos, the movement of the gripper relative to the cube is noticeable, though they are performing the same trajectory file. I am wondering whether this discrepancy is caused by my modifications or by simulators. The difference between mujoco and isaaclab is less noticeable.

geng-haoran, 2025-08-24:
Thank you so much! We are checking this issue!

geng-haoran, 2025-08-25:
After a test, the script works fine on my side. However, i also observe the difference between mujoco and genesis. On my test, genesis can grasp the object but not stable. Another thing to add is that, we are actively developing a new version of metasim, see dev/new_metasim branch. Things are more stable there. We will merge to main branch in a few days. Thank you again for all the information and the great issue. We will also double check the code and find out why the dynamics seems different.