RoboVerseOrg/RoboVerse #457: [Bug Report] Planning failed with cuda errors

boshi-an, 2025-08-30:
### get_started/motion_planning/0_franka_planning.py fails with cuda errors

When running `python get_started/motion_planning/0_franka_planning.py --sim mujoco`

I get the following trace:

```text
[08/31/25 01:39:45] DEBUG    Step 0                                                                                           0_franka_planning.py:150
                    INFO     gripper_out: tensor([ 0.,  0., -1.]), gripper_long: tensor([0., 1., 0.]), gripper_short:         0_franka_planning.py:179
                             tensor([1., 0., 0.])                                                                                                     
/home/boshi/Documents/Science/RoboVerse/get_started/motion_planning/0_franka_planning.py:196: UserWarning: To copy construct from a tensor, it is recommended to use sourceTensor.clone().detach() or sourceTensor.clone().detach().requires_grad_(True), rather than torch.tensor(sourceTensor).
  ee_pos_target[0] = torch.tensor(position, device="cuda:0")
Traceback (most recent call last):
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/particle/particle_opt_base.py", line 241, in _initialize_cuda_graph
    self._cu_act_seq = self._run_opt_iters(self._cu_act_in, shift_steps=shift_steps)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/particle/particle_opt_base.py", line 255, in _run_opt_iters
    self.update_seed(init_act)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/particle/parallel_mppi.py", line 367, in update_seed
    self.update_init_mean(init_act)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/particle/parallel_mppi.py", line 374, in update_init_mean
    if not copy_tensor(init_mean, self.mean_action):
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/util/tensor_util.py", line 32, in copy_tensor
    mem_tensor.copy_(new_tensor)
RuntimeError: CUDA error: operation not permitted when stream is capturing
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/boshi/Documents/Science/RoboVerse/get_started/motion_planning/0_franka_planning.py", line 199, in <module>
    obs = move_to_pose(
  File "/home/boshi/Documents/Science/RoboVerse/get_started/motion_planning/0_franka_planning.py", line 129, in move_to_pose
    result = robot_ik.solve_batch(Pose(ee_pos_target, ee_quat_target), seed_config=seed_config)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/wrap/reacher/ik_solver.py", line 823, in solve_batch
    return self._solve_from_solve_state(
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/wrap/reacher/ik_solver.py", line 1074, in _solve_from_solve_state
    result = self.solver.solve(goal_buffer, coord_position_seed)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/wrap/wrap_base.py", line 155, in solve
    act_seq = self.optimize(seed, shift_steps=0)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/wrap/wrap_base.py", line 77, in optimize
    act_seq = opt.optimize(act_seq, shift_steps)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/opt_base.py", line 171, in optimize
    out = self._optimize(opt_tensor, shift_steps, n_iters)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/particle/particle_opt_base.py", line 210, in _optimize
    self._initialize_cuda_graph(init_act.clone(), shift_steps=shift_steps)
  File "/home/boshi/Documents/Science/RoboVerse/third_party/curobo/src/curobo/opt/particle/particle_opt_base.py", line 240, in _initialize_cuda_graph
    with torch.cuda.graph(self.cu_opt_graph, stream=s):
  File "/home/boshi/.local/lib/python3.10/site-packages/torch/cuda/graphs.py", line 186, in __exit__
    self.cuda_graph.capture_end()
  File "/home/boshi/.local/lib/python3.10/site-packages/torch/cuda/graphs.py", line 84, in capture_end
    super().capture_end()
RuntimeError: CUDA error: operation failed due to a previous error during capture
CUDA kernel errors might be asynchronously reported at some other API call, so the stacktrace below might be incorrect.
For debugging consider passing CUDA_LAUNCH_BLOCKING=1
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

[1]    364282 segmentation fault  python get_started/motion_planning/0_franka_planning.py --sim mujoco
```

### System Info

- Commit: [b61b74d]
- Simulator Backend Version: Any
- OS: [Ubuntu 20.04]
- GPU: [GTX 1050 Ti]
- CUDA: [12.4]
- GPU Driver: [550.114.03]

boshi-an, 2025-08-30:
The issue is consistent on my system, with backends including sapien2, sapien3, mujoco.

MurphyZhao04, 2025-09-25:
This error is caused by CUDA/driver issues with cuRobo’s graph capture. We’ve added **PyRoKi**, which is easier to set up and avoids these CUDA problems. Please switch to PyRoKi for planning/IK.

closed 2025-09-25 by MurphyZhao04